goecharger/go-eCharger-API-v2 #257: how to set battery consumption manually over the api?

gocarlos, 2025-01-12:
my go-e controller sensors are connected to the grid (1/2/3) as well as to the PV(5/6/7)

how i also want to connect my AC Battery which is not near the go-e controller therefore need to be connected over API

i've seen that i could do:

```sh
# battery feeding in 1000w
curl http://192.168.0.54/api/set?ids={"pGrid":0,"pPv":0,"pAkku":1000}
```

i receive however 

```txt
nullnullnull
```

and the overview of the controller still shows the battery in a greyed colour (keine Batterie)

how can i tell the controller about the existence of the battery so that the controller can prfioritize the battery when doing PV-Surplus (at the moment sometimes it starts getting power from the battery even though i set to only start getting power after 1000w surplus)

schinagl, 2025-01-12:
You have to do 'ids' every 5 seconds. If you read the results then within 10 seconds from issuing, then the just written results are properly returned

gocarlos, 2025-01-12:
Thanks for the hint, i tried to create a minimal viable script to test your suggestion:

```sh
counter=0
while true; do
    counter=$((counter + 1))
    timestamp=$(date -u +"%Y-%m-%dT%H:%M:%SZ")
    echo "Iteration: $counter / Timestamp: $timestamp"
    curl -H 'Content-Type: application/json' http://192.168.0.54/api/set?ids={"pGrid":0,"pPv":0,"pAkku":1000} 
    curl -H 'Content-Type: application/json' http://192.168.0.54/api/status?filter=pAkku
    echo ""
    sleep 5
done
```

i used the keys described in here: https://github.com/goecharger/go-eCharger-API-v2/blob/main/apikeys-de.md?plain=1#L123 

unfortunatly, still the same behaviour:

[image]

I'm using the latest stable version of the controller: 1.1.1

erxbout, 2025-01-16:
The controller accepts an array of external powers in the 'ecp' api key

To be futureproof use the 'ccn' api key to read the category names and use the index from the category you want to write for the 'ecp' array..

If you send null, nothing changes.. so result would look like

_ccn response_
``` json
[
    "Home",
    "Grid",
    "Car",
    "Relais",
    "Solar",
    "Akku",
    "chargertest",
    "Hall",
    "Garden",
    "Custom 4",
    "Custom 5",
    "Custom 6",
    "Custom 7",
    "Custom 8",
    "Custom 9",
    "Custom 10"
]
```

_ecp to send.. Index of ("Akku") -> 5_
``` json
[
    null,
    null,
    null,
    null,
    null,
    100,
    null,
    null,
    null,
    null,
    null,
    null,
    null,
    null,
    null,
    null
]

```