goecharger/go-eCharger-API-v2 #258: MQTT password problem

MajoV, 2025-01-16:
It is not possible to enter an MQTT password containing the @ character.
This is a huge problem for us because our MQTT broker password contains this character.

Is there any solution for this?

erxbout, 2025-01-16:
Can confirm this.

When setting 'mcu' (MqttBrokerUrl) to something like that:
mqtt://userWithSpecialPassword:r@kz9QN3fsteJH4S@192.168.1.93:1883

The parser throws an error and does not save the change on 'mcu'

However, when you url escape the '@' character it is accepted by the charger..
```
mqtt://userWithSpecialPassword:r%40kz9QN3fsteJH4S@192.168.1.93:1883
```

In the end this still does **not** help, because the library we use does not url decode the password and sends the character as is ('%40')

We opened an issue regarding the bug there:
https://github.com/espressif/esp-protocols/issues/734

theNorr, 2025-11-26:
So that's the issue I'm having. Although in my case it's regarding a password with Swedish ÅÄÖ in it.