goecharger/go-eCharger-API-v2 #260: Detect Cheap Awattar loading

schinagl, 2025-02-03:
Hi 

Zwei Fragen/Anregungen

- Gibts eine verlässliche Art draufzukommen, wann das geschedulte, Daily Trip Laden gerade stattfindet? Eine Mischung aus car, lmo, modelStatus und alw? (I werd nämli net schlau...)
Warum: Ich möchte zu dem Zeitpunkt den HausAkku auf e.g. 200w begrenzen, weil mir sonst das Auto den Hausakku ausleert

- Schön wär überhaupt draufzukommen, wann die Wallbox das billige daily Trip laden scheduled. Damit könnte man obiges Problem lösen, und auch andere 'Cheaploader' koordinieren, die sonst in den gleichen Zeitslot fallen, wie die go-e

Erfahrungen, Ideen? Bin für jeden Tipp dankbar

0xFEEDC0DE64, 2025-02-03:
der modelStatus sollte während den situationen einen entsprechenden Grund angeben.

Der Charger weiß nicht vorab schon, wann er was machen wird.

Es wird jede sekunde von grund auf neu berechnet, ob er gerade laden darf und wie viel. Den Grund gibt er dann mit an mit dem modelStatus.

Die folgenden modelStatus Werte könnten interessant sein

ChargingBecauseAwattarPriceLow=7

ChargingBecauseAutomaticStopTestLadung=8,
ChargingBecauseAutomaticStopNotEnoughTime=9,
ChargingBecauseAutomaticStop=10,

ChargingBecausePvSurplus=12,

schinagl, 2025-02-03:
Ah, Danke!
 Hmm ja😉
Ich hab natürlich schon eine Nacht mitrekorden von modelStatus hinter mir. Daher:
Bei Daily Trip mit stündlichen Tarif geht der Modelstatus auf 10, wenn der Charger zieht. 
Soweit so gut. 

Aber: er geht danach auf
18, und danach wieder auf 10 aber er zieht nix. 

Das kommt daher: 
Ich sag ihm 40kWh bis 8:00, aber der Akku war bei 36kwh voll. Daher geht er für kurze Zeit auf 18, wenn er die 36 hat. Dann wieder auf 10, obwohl kein Strom fließt und wartet. Ich hätte mir ja gedacht er bleibt in 18, aber leider. 

Somit ist 10 nicht eindeutig. 

Aber: ich werd das mit nrg verknüpfen: 
If lmo == 5 && modelStatus == 10 und nrg.p != 0 then
<begrenze die Leistung aus dem hausakku>

schinagl, 2025-02-07:
### Weiter analysiert:
[image]
[image]
[image]

### Problem
Bis 0800 sollen 40kWh geladen werden. Das Auto ist um ca 02:20 fertig und der modelStatus bleibt auf ChargingBecauseAutomaticStop == 10 bis 05:30
Ich rate nun mal, dass die goe, sich den günstigsten Ladebeginn um 3:00 ausrechnet (ohne den geht es nicht), und da sie nicht wissen kann, mit welcher Leistung sie laden wird ‚zieht sie sich zu warm an', und schätzt dass sie mit 8kw laden kann.

Um 40kWh mit den Randbedingungen unterzubringen, beginnt sie um 0:00, und glaubt bis 05:30 (inkl 30 min Pause um 00:38) fertig zu sein. Soweit so gut. Das würd ich auch so machen. (Naja die halbstündige Pause um 00:38 versteh ich nicht, aber egal)

### Conclusio
- Ich würde den modelStatus nicht bis zum Ende der theoretischen Ladezeit um 05:30 auf ChargingBecauseAutomaticStop lassen, sondern nach dem Laden um 02:30 wieder auf 18 geben, denn das tut sie in der LadePause von 00:38 bis 01:08 auch
- Es wär schön, wenn ich die erwartete LadeLeistung angeben könnte, denn dann braucht sie sich nicht ‚zu warm anziehen‘ und schon um 0:00 beginnen. Wenn ich 10kW bei 40kWh vorgeben könnte, würde ich 4 Stunden von 01:00 bis 05:00 laden

schinagl, 2025-02-09:
### Weiter gehts
Verbessernd wirkt die automatische Phaseumschaltlung zu enablen. Dann schaut E_Car(via wh)  und P_Car (via nrg[11]) so aus
[image]
Wenn E_Car das Integral von P_Car ist, und beides von der goe kommt, dann geht das pyhsikalisch nicht: E_Car steigt nur während mit 11kW geladen wird, danach wird mit 4kW geladen, und E_Car wird nicht mehr größer.

GIbts da eine Erklärung? Kann sein, dass die goe nach der Phaseumschaltung auf 1 Phase nicht sauber weiter integriert?

Ah ja und der Strom *wird* vom Grid gezogen
[image]
Also die 8kWh gibts wirklich.

Hat wer Ideen? @0xFEEDC0DE64 ?