goecharger/go-eCharger-API-v2 #271: go-e Charger bleibt stecken bei "Verbinde mit deinem Auto" (56.8 und 57.1)

radiohead3, 2025-05-01:
Hallo zusammen,

Habe meine Go-e Charger Gemini mit Loxone verbunden und hat bisher immer super verbunden. Habe es über HTTP Befehle gelöst. 

Diese kommen auch noch immer sauber an. Wenn die Sonne rauskommt wird die Ladegeschwindigkeit in der go-e Charger App sichtbar hochgeregelt. (Aktuell bei 13A mit 3 Phasen)

Jedoch bringe ich seit ca. zwei Tagen die go-e Charger Gemini nicht mehr dazu, wirklich zu laden. Dies obwohl das Auto angesteckt ist und es Platz hätte im Akku (70%).

[image]

Kennt jemand das Problem?
Hatte zuvor 57.1 installiert aber nun auf 56.8 zurückgestellt. Funktioniert bei beidem nicht mehr.

chk2902, 2025-05-01:
"modelStatus" (HTTP wert) anschauen (https://github.com/goecharger/go-eCharger-API-v2/blob/main/apikeys-de.md)

... und mal die Kontakte der Stecker und Buchsen am Typ2-Kabel, Auto und Wallbox mit Kontaktspray reinigen

radiohead3, 2025-05-01:
Modelstatus 15 (wenn ich http://IP Adresse/api/status eingebe im Browser)

Was heisst das genau?

chk2902, 2025-05-01:
ChargingBecauseFallbackDefault -> eigentlich sollte er laden (interpretiere ich so). Schau mal "alw" an, wenn das "true" ist, dann checke mal die Hardware (Kabel etc).

radiohead3, 2025-05-01:
Ok. Beim öffentlichen Ladepunkt geht es auch nicht mehr. Somit zu 99% ein Problem bei unserem Fahrzeug. Werde morgen mal in der Garage vorbeigehen.

closed 2025-05-01 by radiohead3

chk2902, 2025-05-01:
... oder das Ladekabel von Dir. Oder die Typ2-Buchse am Auto.

chk2902, 2025-05-01:
Hätte sein können, dass jeder mit seinem Kabel geladen hat. Ja, da hast Du recht. Also Auto untersuchen lassen-

radiohead3, 2025-05-01:
Werde ich machen! Vielen Dank