goecharger/go-eCharger-API-v2 #276: some API Keys not working via MQTT

baycom, 2025-07-09:
I think with the latest firmware version 59.4 the API keys 'ids', 'pGrid', 'pPv'' and 'pAkku' are not working via MQTT yet. I tried sending these messages:
Topic: go-eCharger/XXXXXX/ids
Message: {"pGrid":1000}

Topic: go-eCharger/XXXXXX/pGrid
Message: 1000

For both API keys / messages there are no results coming back and readout via HTTPv2 API does not show any change (http://x.x.x.x/api/status?filter=pgrid). 

The HTTPv2 API works as expected:
http://x.x.x.x/api/set?ids={%22pGrid%22:1000,%22pPv%22:2000,%22pAkku%22:0}

Did I do something wrong or is this feature missing the firmware by accident or on purpose?

Best,

Deti

ksmolder, 2025-07-23:
Did you try to publish to the topic `go-eCharger/XXXXXX/ids/set` as well? For me this works using firmware 59.4.

baycom, 2025-07-23:
Yes, tried that too, but cannot read back any value set via http API: http://xxx.xxx.xxx.xxx/api/status?filter=ppv,pakku,pgrid
Maybe you could give an explicit example on how the JSON message should look like?