goecharger/go-eCharger-API-v2 #283: PV-Überschussladen ohne PV-Akku leerzusaugen?

fredlcore, 2025-09-16:
Erst mal ein großes Dankeschön an go-e, dass hier so eine umfangreiche API zur Verfügung gestellt wird.
Meine Konfiguration: 8kWp PV, 5kW Solaredge Wechselrichter, 8kWh Byd Battery Box LVS, 2 go-e Home+ Wallboxen.

Ich sende alle fünf Sekunden pGrid, pPv und pAkku an die go-e und kann von daher die PV-Überschussfunktion in der App nutzen und aktivieren. Ab 1.4kW Überschuss soll geladen werden, Leistungspräferenz ist auf "Strom ins Netz bevorzugen". 
An Tagen wie heute, wo zeitweilig weniger als 1.4kW produziert werden, pausiert der Ladevorgang aber nicht, sondern die Differenz wird aus der (zwischenzeitlich geladenen) Byd Battery Box gezogen. Das ist nicht Sinn der Sache, weil das Auto zwar prioritär geladen werden soll, aber nicht auf Kosten des Akkus.

Ich frage mich, wofür pAkku übermittelt wird und ob ich hier einen Fehler habe: Wenn aus der Byd Strom gezogen wird, übermittle ich einen positiven Wert für pAkku, wenn die Batterie aufgeladen wird, wird ein negativer Wert übermittelt.

Wenn über 1,4kW PV-Übeschuss vorhanden sind, scheint es zu funktionieren, aber wenn einmal die 1,4kW überschritten wurden und dann zwischenzeitlich abfallen, wird anscheinend aus der Byd-Batterie ausgeglichen, um auf 1,4kW zu kommen. Ließe sich das irgendwie verhindern, so dass beim Unterschreiten der 1,4kW pausiert wird und nicht aus der Byd-Batterie gespeist wird?

schinagl, 2025-09-16:
> Wenn über 1,4kW PV-Übeschuss vorhanden sind, scheint es zu funktionieren, aber wenn einmal die 1,4kW überschritten wurden und dann zwischenzeitlich abfallen, wird anscheinend aus der Byd-Batterie [...]

Das kannst nicht verhindern weil du sonst dauernd aus und ein des Ladens hast. 

Das wurde mal 'wolkensmoothing' von einem Go-e Entwickler genannt

chk2902, 2025-09-16:
Ich nenne das bei meiner App (Wallbox-Steuerung für Android) Ein-/Ausschaltverzögerung oder Wolkenüberbrückung. Man sollte eigentlich einstellen können, wie lange dieser "Nachlauf" aktiv ist, nach dieser Zeit (so 5 Minuten soder so sind sinnvoll) sollte das dann abschalten.

chk2902, 2025-09-17:
Das wäre nicht gut. Klar, der Nachlauf muss auch bei leerem Akku gemacht werden, dann halt von externem Bezug. Besser als dass irgendwann mal der Schütz im Auto - jedes Bauteil hat eine begrenzte Funktion - aufgibt. Aber so was finde ich sollte man einstellen können...

chk2902, 2025-09-17:
Ich verwende meine App - die ist billiger als der Controller und komfortabler. Ich wollte nur den Grund für einen solchen Nachlauf erläutern. Auch bei Phasenumschaltung sollten Hysteresetoleranzen im Minutenbereich dabei sein, sonst ist bei Bewölkung nur noch Trommelwirbel in der Garage ;)

schinagl, 2025-09-17:
Ich mach das auch mit Python und noch so manches mehr damit. 

Aber die Frage oben war an @fredlcore

fredlcore, 2025-09-17:
@chk2902: Ich bin kein Elektriker, aber macht es für das Auto einen Unterschied ob nun minimal oder gar kein Strom kommt? Die Wallbox moduliert ja ansonsten auch im 0,1A-Bereich, zumindest zeigt die go-e App das so an. Phasenumschaltung kommt bei mir nicht vor, weil der Leaf nur einphasig laden kann.

@schinagl: Ich nutze die App zum Starten des Eco-Ladens und um dort einzustellen, dass nur PV-Überschussladung genutzt werden soll. Ansonsten verwende ich FHEM und sende alle fünf Sekunden das hier:
`http://192.168.1.45/api/set?ids={%22pGrid%22:%%Meter_Power%%,%22pAkku%22:%%Battery_Instantaneous_Power%%,%22pPv%22:%%pvProduction%%}`
Wobei `Meter_Power` negativ ist, wenn ich einspeise, `Battery_Instantaneous_Power` negativ ist, wenn der Akku über PV aufgeladen wird und `pvProduction` positiv ist, wenn PV-Überschuss vorhanden ist.

Ihr könnt also bestätigen, dass `pAkku` berücksichtigt wird? Sprich, dass die Ladeleistung des EV (zumindest nach einigen Minuten) gedrosselt wird, wenn während des Ladens auch die Hausbatterie in Anspruch genommen wird? Dann muss ich das noch mal genauer beobachten bzw. mal die FHEM-Logs analysieren...

chk2902, 2025-09-17:
Ein go-eCharger regelt nur in ganzen A-Schritten (leider, der Leaf kommt auch gut mit feineren Schritten zurecht, Tesla nicht, das macht da wieder ganze A draus). Daher hängt der Leaf bei mir an einer SmartWB, da kann man noch ein paar Prozent mehr rauskitzeln bei der besseren Nachführung an die Überschussleistung ;)

Aber unter 6 A pro Phase kannst Du überhaupt nicht gehen. Kann m.W. keine Wallbox. Auch wenn das Protokoll das erlauben würde - die Lade-Verluste steigen zu stark an, und das ist zu einem Quasi-Standard geworden.

schinagl, 2025-09-18:
Wie pgt wenn das laden neu startest? 

Es gilt nämlich ganz allgemein:

PPV + PAkku + PGrid + pgt > 1.4

Um zu laden. 

Pgt ist Power Grid Target und lässt sich in der App via 

Eco -> Ziel Wert Netz

Einstellen. Programmatisch natürlich auch über http api.

fredlcore, 2025-09-18:
Das kommt höchstens 2-3 Mal pro Stunde vor, und zu den Aussetzer-Zeiten geht die Wallbox dann ja auch in die Unterbrechung (weil die Werte dann eben 0 sind). Wenn ich in der App dann gerade in den Eco-Einstellungen bin, erscheint kurz der Hinweis, dass kein go-E-Controller vorhanden ist, der verschwindet aber sofort von selber wieder, wenn die Werte nach 1-2 Iterationen wieder eintrudeln. Von daher ist also das Problem aus den Ausbleibenden Übermittlungen höchstens, dass dann die Ladung unterbricht, obwohl theoretisch weiterhin Überschuss vorhanden ist.

Das eigentliche Problem bleibt ja aber, dass über längere Zeiträume die von Dir genannte Formel nicht erfüllt ist und trotzdem geladen wird.

schinagl, 2025-09-18:
Hmmm... Du hast Firmware 59.4?

Das eigenartige ist ja dass das bei mir auf zwei Installationen funktioniert. 
Und auch im Forum machen das viele ohne go-e Controller und eben via Home Assistent oder wie ich mit raspi und Python. 

Also das ist ziemlich sicher kein grundsätzliches Problem der go-e. Das gäbe den Mega Aufschrei hier. 

Bin auch am Ende meiner Weisheit

fredlcore, 2025-09-18:
Firmware ist bei mir übrigens die 57.1, was anderes wird in der App (für den Go-E Home+) nicht angeboten...

fredlcore, 2025-09-18:
Nach einem Neustart der Wallbox richtet sich das Laden entsprechend der Werte, die pGrid übermittelt. Wenn der Export unterhalb der Ladeschwelle liegen, wird nicht geladen, andernfalls (wenn mehr als 1,4kW exportiert werden) wird geladen.  
**Allerdings wird nach dem ersten Starten des Ladens nicht mehr pausiert, wenn der Wert dann wieder unter die nötigen 1,4kW fällt!**  
Ich habe jetzt heute Abend die Werte, die FHEM sendet, fest eingetragen und zwischen 1000, und -1500 wechseln lassen. Wenn ich nach dem Reset mit 1000 starte (also Strom aus dem Netz beziehe), wartet die Wallbox korrekterweise mit dem Laden. Wenn ich dann -1500 sende, fängt sie an zu laden. Wenn ich dann allerdings wieder auf 1000 zurück gehe, lädt sie weiterhin, und zwar immer mit minimaler (also 6A) Leistung, egal, wie gering meine Produktion oder wie hoch mein Strombezug ist. Erst nach einem erneuten Reset geht es dann wieder.
Leider finde ich meine Werksreset-Karte nicht, um zu testen, ob es einfach eine verkorkste Einstellung ist, aber so viel kann man ja im Eco-Modus nicht einstellen. Es ist mir echt ein Rätsel...

fredlcore, 2025-09-18:
Ich habe das gleiche nun auch an unserer zweiten go-e Wallbox getestet, gleiche Firmware, gleiches Verhalten. Auch ein Downgrade auf ältere Firmwares, z.B. 55.8, hat nichts verändert.
Hier noch ein Log, bei dem man sehr gut sehen kann, welche Auswirkungen die simulierten pGrid-Werte (3. Spalte) von 1500 (Netzbezug) zu -1500 (Export über der 1,4kW-Schwelle) und dann wieder zu 2500 (Netzbezug):

[log.csv](https://github.com/user-attachments/files/22416196/log.csv)

Beim Wechsel von -1500 zu 2500 sieht man, wie in der 8. Spalte `cll.pvSurplus` von 16 Ampere über 9A auf 6A fällt, aber dort dann verharrt. Nur nach einem Neustart wartet die Wallbox brav so lange, bis wieder ein Überschuss vorhanden ist, verfällt dann aber wieder in das gleiche Muster.

fredlcore, 2025-09-21:
Und hier das gleiche noch einmal nicht mit simulierten Werten, sondern die Situation von heute morgen, nachdem ich einen Neustart ausgeführt habe:

08:00:21
- Erster Eintrag nach Neustart, kein PV-Überschuss, `modelStatus` geht nach ein paar Sekunden auf 17 (`NotChargingBecauseFallbackAwattar`), `car` ist auf 3 (`WaitCar`), cll.pvSurplus ist leer (weil ja auch kein Überschuss vorhanden ist). Es erfolgt keine Ladung, korrekt.
- Die bestehende PV-Produktion (`ppv`) liegt bei ca. 600 Watt und geht zu einem guten Teil in den Akku (negativer `pAkku`-Wert)

09:20:26
- In der App ist als Mindestüberschuss 2000W eingestellt, diese sind jetzt erreicht und das Auto fängt mit 6A an zu laden und geht kurz danach auf 13A hoch und moduliert ab da entsprechend des PV-Überschusses, `modelStatus` geht auf 12 (`ChargingBecausePvSurplus`). Auch alles noch korrekt.

10:11:53
- Die PV-Produktion fällt auf unter 1,4kW (der eigentliche Überschuss schon davor) und bleibt mindestens bis 10:17:48 unterhalb dieser Schwelle. Spätestens hier (wenn nicht schon beim Unterschreiten der in der App eingestellten 2000W) müsste also das Laden unterbrochen werden. Stattdessen wird die Differenz zu 1,4kW über den inzwischen teilweise geladenen PV-Akku geliefert. `modelStatus` bleibt auf 12.

11:25:18
- Jetzt ist der Akku entladen und kann nichts mehr liefern, PV ist wegen Regens inzwischen minimal, aber es wird weiter geladen, nun mit komplettem Bezug aus dem Netz. `modelStatus` bleibt auf 12.

11:30:11
- Jetzt habe ich den Ladestecker abgezogen, und `car` geht von 2 (`Charging`) auf 1 (`Idle`). `modelStatus` bleibt auf 12.

Wie gesagt, das lässt sich an beiden Wallboxen reproduzieren. Die einzige Möglichkeit, die Ladung zu unterbrechen, ist, wenn keine PV-Werte übermittelt werden (sieht man an den hin und wieder auftauchenden Zeilen, wo ppv, pakku und pgrid leer sind. Dann geht die App in den Status, dass zu wenig PV-Überschuss vorhanden ist und pausiert die Ladung, bis wieder Werte kommen.

[ChargingLog.csv](https://github.com/user-attachments/files/22451052/ChargingLog.csv)

fredlcore, 2025-09-21:
So, ich habe jetzt ein Script geschrieben, das kontinuierlich auf diese Bedingung prüft:
```
        pvopt_averagePPv < 1400
        and pakku >= 0
        and pvopt_averagePGrid > 0
        and cll.pvSurplus == 6
        and car == 2
```
Wenn das zutrifft, wird über `/api/set?rst=1` ein Reset der entsprechenden Wallbox durchgeführt. Dadurch, dass zwei Average-Werte verwendet werden, nutze ich weiterhin das "Wolken-Aliasing" der Wallbox, und es kommt nicht bei einem kurzzeitigen Unter-/Überschreiten der Werte zu einem Neustart. Der Reboot dauert nur wenige Sekunden, so dass er nicht weiter ins Gewicht fällt. Schön ist was Anderes, aber es läuft:

Ich habe nun mehrfach heute Abend zuerst mit simulierten Werten einen deutlichen PV-Überschuss vorgegaukelt, damit das Laden beginnt, und sobald ich dann die realen Nacht-Werte aus dem Wechselrichter übermittle und die Bedingungen zutreffen, wird ein Reset ausgelöst. Dann wartet die Wallbox brav mit dem Laden, weil es ja keinen PV-Überschuss gibt. 

Falls der Support hier mitliest: Da das Problem nach einem Neustart (erst einmal) verschwindet, gehe ich nicht von einem Konfigurationsfehler aus. Was mir aufgefallen ist, ist, dass sowohl nach einem Neustart, als auch dann, wenn keine pGrid-Werte übermittelt werden, cll.pvSurplus nicht Teil von /api/status ist. Der Wert verschwindet ohne meinen Reset-Workaround aber nie von selbst, sondern bleibt eben auch dann auf 6 (Ampere) stehen, wenn es wie jetzt mitten in der Nacht ist. Meine Vermutung ist, dass dieser (oder ein damit zusammenhängender) Wert bei einem Wegfall des PV-Überschusses nicht korrekt gelöscht wird, und möglicherweise auf diesen Wert hin geprüft wird, ob weiterhin PV-Überschuss vorhanden ist. Da cll nicht in der API erklärt wird, weiß ich nicht, ob es einfacher wäre, diesen einfach direkt regelmäßig zu löschen oder auf 0 zu setzen, um damit eine Neuberechnung zu erzwingen. Aber am einfachsten wäre es sicherlich, wenn dieser Bug gefixt würde, auch wenn mich wundert, dass ich scheinbar der einzige bin, der dieses Problem hat ;)...

GuenterPreinfalk, 2025-11-05:
ich spiele da auch eher mit dem Zielwert ins Netz, bei Minuswerten lädt dann der Speicher noch, bei Plus zieht er aus den Speicher oder eben aus dem Netz